Team2890HawkCollective/Quetzalcoatl #8: Program motors for Intake mechanism

Team2890, 2019-01-15:


Podcakero, 2019-01-18:
Most probably will be driven by Talons, consisting of AT MOST 3 redline motors, each which needs to be separately controlled. For intake, bottom two are spun up in opposing directions. For outake, front two will be spun up in opposite directions (for a 3 motor system).
For 2 motor: intake - rear motor spun up, front stays stationary. Outake - front motors spins up, rear stays stationary
Motors spin fast, so have a modifier of like... 0.05 (5%) max motor speed.

Podcakero, 2019-01-18:
Configuration of motors not finalised

Podcakero, 2019-01-20:
It is decided (currently) that one motor will be used for intake/outake
And a servo at the top for keeping the ball in/letting it out
See: Ty/Slater

Valance23322, 2019-01-25:
Servo is being re-purposed to control the hatches.  Blocked while waiting on hardware implementation.

Podcakero, 2019-02-05:
we are awaiting finalisation of cad and the prototype to be finished

Valance23322, 2019-02-05:
Preliminary code has been written, currently awaiting hardware fixes so code can be tested.

Valance23322, 2019-02-19:
Code has been tested and correctly spins the motors, however the mechanism does not pick up the ball.  Team is currently attempting hardware modification to resolve issue.

Team2890, 2019-02-27:
Mechanism picks up and shoots balls.
can pick up from bottom and front

closed 2019-02-27 by Team2890