Team2890HawkCollective/Quetzalcoatl #40: Configure buttons for autonomous control

Team2890, 2019-02-12:


Team2890, 2019-02-13:
Created [Automated Subsystems.](https://github.com/Team2890HawkCollective/Quetzalcoatl/blob/dfcc57f6d71c227edfe8f1dc63c10261ac3a7006/src/main/java/frc/robot/subsystems/AutomatedSubsytem.java)([commit](https://github.com/Team2890HawkCollective/Quetzalcoatl/commit/dfcc57f6d71c227edfe8f1dc63c10261ac3a7006)) In there, added buttons A, B, and Y to the Assisted Driver's Controller. The buttons move the lift to the height of the different levels. Changed it to an enumerated type called Rocket Level 1, 2, and 3. Implemented javadoc for the method. Added automated subsystems to RobotMap. Implemented xboxAutonomousControl to xboxDriveCommand and joystickDriveCommand.

Team2890, 2019-02-27:
The buttons to control elevator movement work
Need to test pixy cam

Podcakero, 2019-02-27:
Buttons are configured
And PixyCam testing is taken care of by #33

closed 2019-02-27 by Podcakero