Team2890HawkCollective/Quetzalcoatl #48: RaiseElevatorCommand will always move the elevator up, even if we need to go down

Podcakero, 2019-02-16:
Change name to MoveElevatorCommand and have a check to see if the current position if less than or greater than the target, and call elevatorUp or elevatorDown accordingly

Team2890, 2019-02-18:
Branched from master into new branch ElevatorCommand. Changed name to MoveElevatorCommand and checked whether or not position of encoder was equal to, greater then, or less than the encoder position. elevatorUp and elevatorDown are called accordingly. Needs testing

Valance23322, 2019-02-19:
Changes have been merged into master, this did not work at last test

Team2890, 2019-02-27:
Last test this did allow the elevator to move between levels with ease, even from level 3 to level 1

closed 2019-02-27 by Team2890